FrameworkComputer/InputModules #1: Trackpoint please, please *please*

chx, 2023-03-24:


jyancat, 2023-03-26:
Issues are not for feature requests, this is generally for reporting issues with anything in this repository. If you want to request a feature/new module you can use the forums [https://community.frame.work/](url)

eclecticc, 2023-04-15:
@jyancat 's comment is correct.  Closing out.

closed 2023-04-15 by eclecticc

chx, 2023-04-15:
Extremely disappointing.

eclecticc, 2023-04-15:
To be clear, @jyancat 's correct comment is that GitHub Issues are for reporting bugs or adding enhancements within the code or CAD that exists in the repository.  Feature requests for the product itself go in the Framework Community.

cscott, 2024-01-23:
See https://community.frame.work/t/thinkpad-keyboard-mod-super-early-stages/38484 for active discussion.