FrameworkComputer/InputModules #2: compatible key switches

stuart-warren, 2023-03-24:
I'm very excited by this project, especially the usb support.

If we were to build homebrew input modules that are more ergonomic what key-switches would even fit under the screen?

Kailh choc switches look like they may not, and I don't believe cherry mx ultra low profile are even available to the public in small quantities.

Ideally I'd like to build (assuming you won't provide) an input module in a similar style to 
* https://golem.hu/board/Choctopus44-439.html
* https://golem.hu/board/Zephirum-440.html

carcinization, 2023-03-24:
the question i have is more "what parts are going to be used in the keyboard it deploys with?" because if they open up that information, that will make it easier for us to just remix that design without worrying about sourcing new parts to fit and reinventing the wheel. obviously the keyboard it deploys with probably won't be mechanical, but I figure it's more important to have flexibility of layout than having the actual mechanical switches. plus then we could re-use the keycaps from the original keyboard module!

eclecticc, 2023-04-15:
Closing this since it isn't addressable within this repository.  There is ongoing discussion around keyswitches in the Framework Community: https://community.frame.work/t/input-module-keyboard-switches/28294/8

closed 2023-04-15 by eclecticc