FrameworkComputer/InputModules #3: SK6812MINI-E Logic Voltage

jyancat, 2023-07-22:
In [https://github.com/FrameworkComputer/InputModules/tree/main/Electrical/MicrocontrollerInputModule](url) there is a SK6812MINI-E RGB LED driven by 3V3 logic, while this probably will work it's not optimal, better to add a level shifter (up to 5V logic) or note this in the folder readme.

cscott, 2024-01-24:
The datasheet specifies that VIH is "3.4V typical" so yeah, this is technically operating the SK6812 out of spec.
[image]

cscott, 2024-07-10:
A 74LVC1G34GW would be a compact solution.

JohnAZoidberg, 2024-10-13:
> this probably will work

You're right, it does work :D

This really is just a quick and dirty reference that's not supposed to be used for anything and also supposed to be as simple as possible.
We made a couple test boards to see if it works, but nothing more.

So if you could make a PR to add a level shifter that would be great.
Otherwise I think we'll just add a comment to the README noting that the design is functional but perhaps not optimal.