FrameworkComputer/InputModules #6: BOARD_ID resistor definitions

cscott, 2024-01-23:
Under "Pinout", the [README.md](https://github.com/FrameworkComputer/InputModules#pinout) says,
> 4 	BOARD_ID 	Pull to GND through BOARD_ID resistor defined below

but the BOARD_ID resistor values aren't actually defined anywhere in the repo.

I'm particularly interested in knowing how many of the 8 different pogo pin groups need to be connected in order for the BIOS to be happy.  To pick two specific examples: if I created a "double height" keyboard+touchpad module, but left all four side spacers in place, would I need two different pogo pin pads, or could I do with just one and a "this is really two modules" resistor value?

cscott, 2024-01-24:
For anyone wondering, the "microcontroller" example has a 180k resistor from BOARD_ID to ground, so that's apparently the correct value for a "small" module.  No info yet on appropriate values for other size modules.

eclecticc, 2024-01-30:
Added in https://github.com/FrameworkComputer/InputModules/commit/8348517b89d8de535c8a948ec6aa91f2cee6927c

closed 2024-01-30 by eclecticc

cscott, 2024-01-31:
And for my second question above, I'm guessing that if I have a big "keyboard plus touchpad" module I'd need two separate pogo pads with two separate resistors, one for the "keyboard" and another for the "touchpad"?

eclecticc, 2024-01-31:
Yes, as currently defined.  That is something we could define in the future
though.