FrameworkComputer/InputModules #8: Pinout and design files for trackpoint module

cscott, 2024-01-31:
I'm working on a "double-height" keyboard module that will replace both the keyboard and the touchpad module.  [We don't have any documentation on the touchpad module](https://community.frame.work/t/mechanical-electrical-specs-for-touchpad-module/40833) other than that it has a [different pinout than the other modules](https://community.frame.work/t/input-module-placement-qs/34494/2) and [the board ID resistor value](https://github.com/FrameworkComputer/InputModules/issues/6).

Even just knowing where the board ID and GND pins were on the touchpad module (the same as for the input modules? or different?) would help my particular use case, since I just want to assure the FW16 EC that the touchpad is "present".

eclecticc, 2024-02-01:
https://github.com/FrameworkComputer/InputModules/commit/02588e58b65d8cc1fac26795650c8f1eb097ab79

closed 2024-02-01 by eclecticc

cscott, 2024-02-01:
Fantastic!  I think the final image there might be flipped, though, as described here: https://community.frame.work/t/input-module-placement-qs/34494/6