FrameworkComputer/InputModules #10: I2C protocol for touchpad module

cscott, 2024-02-02:
Again, thanks so much for the information provided in #8.

There have been requests to create [alternate touchpads](https://community.frame.work/t/touchpad-with-2-buttons-instead-of-clickpad/36343), and the missing piece to do this (other than #9, the mechanical specs) would be details of the I2C protocol spoken by the touchpad.  There is a "[HID over I2C](https://github.com/torvalds/linux/blob/master/drivers/hid/i2c-hid/i2c-hid-core.c)" thing, and if that's what's going on here than probably just stating so would be enough.  If its a bespoke protocol, then we'd probably need the datasheet for the stock touchpad used in the FW16 so we could emulate it properly.

JohnAZoidberg, 2024-10-13:
I2C is connected to the CPU, so you can do whatever you want, as long as you have an OS driver that supports it.
Our modules do HID over I2C (on Framework 16 but also on Framework 13).
I see you came to the same conclusion already :smiley:  https://community.frame.work/t/touchpad-with-2-buttons-instead-of-clickpad/36343/18?u=daniel_schaefer
Sorry the the late response.

The same I2C bus is also connected to the EC. So that if you boot to your OS without the touchpad connected, the EC will pretend to be a touchpad. This is so that the windows driver won't freak out. See https://github.com/FrameworkComputer/EmbeddedController/commit/9d49389919c36e44e451514b8278b9eb7ee6ed1e

closed 2024-11-20 by amstan

JohnAZoidberg, 2025-10-18:
@cscott I added some more information here: https://github.com/FrameworkComputer/InputModules/pull/21

PR #17 (2024-11-20): README: Document touchpad protocol
Closes #10
commit 3d60be5b (2024-11-20): README: Document touchpad protocol

Closes #10